bkeevil/esp32-cam #2: OV3660 is hard to find in the market, I have OV5640, do you want it?

playGit888, 2019-08-30:
Hi, dear, I found you on this post: https://github.com/espressif/esp32-camera/issues/65
As I said in the title, I can send it to you by Post Office. Which country are you in? 
I have some OV5640. The pins definition are the same as OV2640, which is used on most of ESP32 boards.
To make it work, you need to adjust voltage for it on your board. I also can send you some resister or LDO.
They are cheap here in China.
If you can write the driver and share it worldwide, it will be very great. (it is not a must)

I read the PDF, it says like this:
// JPEG, 2592x1944 fixed size, 15fps
// Input clock = 24Mhz, PCLK = 42 MHz
And I read codes in esp32-cam, it seems that the setting is xclk 20Mhz, and pclk10Mhz.
I am not sure ESP32 is fast enough for OV5640.  

The picture from OV5640 should be much better than OV2640 and OV3660.
I didn't test OV5640 yet, I don't know how to write the driver on ESP32.
I tested OV5647 on ARM linux, the image quality is great, less noise than OV2640 and OV3660 on ESP32.
Let me know if you are interested.

Thank you for your great project: bkeevil/esp32-cam

playGit888, 2019-08-30:
Forgot to say, all items are free.

bkeevil, 2019-08-30:
Yeah sure. If you want to send me some samples I'm interested in writing or adapting a driver for the module. Will they fit in the the same slot as the OV2640? I have several ESP32-Cam AI Thinker boards with the OV2640 modules but thats about it.

574 Hart Road, Callander, Ontario, Canada, P0H 1H0

playGit888, 2019-08-30:
yes, it fit.....I have this AI thinker board too.
OK, I will try to find some LDO for you to adjust voltage, I will send to that address after a few days,

bkeevil, 2019-08-30:
Why do I have to adjust the voltage?

I have set up a feature branch for the driver [here](https://github.com/bkeevil/esp32-camera/tree/feature/ov5640)

The initial commit is an almost a verbatim copy of the driver for the OV2640 where things have been renamed and the product ID has been set correctly as per the data sheet.

It probably won't work but it may detect the OV5640 module.

bkeevil, 2019-08-30:
Yeah I took a look at  the datasheet and it doesn't look like it is fast enough. It might still work at a lower resolution. The ESP32 is only really capable of 640x480 or less anyways.

playGit888, 2019-09-17:
Hi, dear...
I have not sent the package yet.
I checked the voltages for OV2640. I guess 1.2v, 2.8v. 3.3v are on the AIthinker board. The 3.3v is directly connected to system power supply.
While OV5640 needs, 1.5v, 2.8v, 2.8v. We can change 1.2v to 1.5v by replacing a LDO.
But we can't change the 3.3v to 2.8v easily, because it is directly connected to the power supply.
Do you have other board?

bkeevil, 2019-09-17:
The only other board I have played around with the is TTGO T-Journal but it doesn't have enough an SPI-RAM chip so it is limited to CIF resolution. I don't think it is worthwhile developing an OV5640 driver until someone produces a board that is compatible with it.

bkeevil, 2019-10-25:
This seems to have stalled so I'm closing it.

closed 2019-10-25 by bkeevil