bkeevil/esp32-cam #3: recursive clone results in permissions error

g8ecj, 2020-01-22:
Looks like some/all of the submodules are in a private part of your git repo as github is asking for a user/password combo.

If I miss out the recursive switch, it connects OK and I get the reppo (but without the submodules)

bkeevil, 2020-01-22:
I'll take a look when I get home. In the mean time the needed repos are available from Espressif's github.

g8ecj, 2020-01-24:
I've manually cloned the 2 submodules (esp-idf-ssd1306 & esp32-camera) but had to copy all the in ssd1306 from Main up a level to get anywhere at all compiling the project. 

I've now hit the following whole load of problems! No idea what the issue is (I could have any number of files missing fro mthe submodules!!)

[ 63%] Building C object esp-idf/main/CMakeFiles/idf_component_main.dir/app_settings.c.obj
/home/g8ecj/esp/esp32-cam/main/app_settings.c: In function 'app_settings_reset':
/home/g8ecj/esp/esp32-cam/main/app_settings.c:36:3: error: unknown type name 'nvs_handle_t'
   nvs_handle_t handle;
   ^
/home/g8ecj/esp/esp32-cam/main/app_settings.c:39:38: error: pointer targets in passing argument 3 of 'nvs_open' differ in signedness [-Werror=pointer-sign]
   if (nvs_open(NVS_KEY,NVS_READWRITE,&handle) == ESP_OK) {
                                      ^
In file included from /home/g8ecj/esp/esp-idf/components/nvs_flash/include/nvs_flash.h:21:0,
                 from /home/g8ecj/esp/esp32-cam/main/app_settings.c:6:
/home/g8ecj/esp/esp-idf/components/nvs_flash/include/nvs.h:109:11: note: expected 'nvs_handle * {aka unsigned int *}' but argument is of type 'int *'
 esp_err_t nvs_open(const char* name, nvs_open_mode open_mode, nvs_handle *out_handle);
           ^
/home/g8ecj/esp/esp32-cam/main/app_settings.c:47:29: error: 'CONFIG_LWIP_LOCAL_HOSTNAME' undeclared (first use in this function)
   strncpy(settings.hostname,CONFIG_LWIP_LOCAL_HOSTNAME,LEN_HOSTNAME);
                             ^
/home/g8ecj/esp/esp32-cam/main/app_settings.c:47:29: note: each undeclared identifier is reported only once for each function it appears in
/home/g8ecj/esp/esp32-cam/main/app_settings.c: In function 'app_settings_save':
/home/g8ecj/esp/esp32-cam/main/app_settings.c:59:3: error: unknown type name 'nvs_handle_t'
   nvs_handle_t handle;
   ^
/home/g8ecj/esp/esp32-cam/main/app_settings.c:62:40: error: pointer targets in passing argument 3 of 'nvs_open' differ in signedness [-Werror=pointer-sign]
   ret = nvs_open(NVS_KEY,NVS_READWRITE,&handle);
                                        ^
In file included from /home/g8ecj/esp/esp-idf/components/nvs_flash/include/nvs_flash.h:21:0,
                 from /home/g8ecj/esp/esp32-cam/main/app_settings.c:6:
/home/g8ecj/esp/esp-idf/components/nvs_flash/include/nvs.h:109:11: note: expected 'nvs_handle * {aka unsigned int *}' but argument is of type 'int *'
 esp_err_t nvs_open(const char* name, nvs_open_mode open_mode, nvs_handle *out_handle);
           ^
/home/g8ecj/esp/esp32-cam/main/app_settings.c: In function 'app_settings_startup':
/home/g8ecj/esp/esp32-cam/main/app_settings.c:80:3: error: unknown type name 'nvs_handle_t'
   nvs_handle_t handle;
   ^
/home/g8ecj/esp/esp32-cam/main/app_settings.c:91:39: error: pointer targets in passing argument 3 of 'nvs_open' differ in signedness [-Werror=pointer-sign]
   ret = nvs_open(NVS_KEY,NVS_READONLY,&handle);
                                       ^
In file included from /home/g8ecj/esp/esp-idf/components/nvs_flash/include/nvs_flash.h:21:0,
                 from /home/g8ecj/esp/esp32-cam/main/app_settings.c:6:
/home/g8ecj/esp/esp-idf/components/nvs_flash/include/nvs.h:109:11: note: expected 'nvs_handle * {aka unsigned int *}' but argument is of type 'int *'
 esp_err_t nvs_open(const char* name, nvs_open_mode open_mode, nvs_handle *out_handle);
           ^
cc1: some warnings being treated as errors
make[3]: *** [esp-idf/main/CMakeFiles/idf_component_main.dir/build.make:91: esp-idf/main/CMakeFiles/idf_component_main.dir/app_settings.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:4304: esp-idf/main/CMakeFiles/idf_component_main.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:152: CMakeFiles/app.dir/rule] Error 2
mak

bkeevil, 2020-01-24:
I just noticed that Espressif has yet to merge my pull request #61 which introduces non-volatile storage (nvs) into their esp32-camera repository. The 'nvs' errors you are reporting pertain to that missing code. This means that you can't use Espressif's version of the esp32-camera driver, you need the one in the feature/nvs branch of my fork.

I see no reason why the recursive checkout procedure would require a username and password as I have no private repositories on Github. Have you tried your own username and password? [Take a look at the .gitmodules file](https://github.com/bkeevil/esp32-cam/blob/master/.gitmodules) in the root directory and notice the urls and branches of the submodules that need to be cloned. You could try to cloning those urls directly from my github and see if it gives a useful error message. 

Could it be prompting you for your password for your own private SSH key? You might try modifying the .gitmodules file to use the https style of url for the repo rather than the ssh style url and then run `git submodule update` to pull them down.

Please let me know if any of this helps.

bkeevil, 2020-01-24:
The pull request referenced above was just merged into the espressif/esp32-camera master branch. I will update .gitmodules and delete that fork

bkeevil, 2020-01-25:
There was a typo in the documentation:

`git clone --recursive https://github.com/bkeevil/esp-cam.git`

should have been:

`git clone --recursive https://github.com/bkeevil/esp32-cam.git`

g8ecj, 2020-01-25:
I started totally from scratch with a new clone of esp-idf and your repo and it's looking cleaner!! I still had to manually clone the components submodules 

```
$ git clone --recursive https://github.com/bkeevil/esp32-cam.git
Cloning into 'esp32-cam'...
remote: Enumerating objects: 423, done.
remote: Counting objects: 100% (423/423), done.
remote: Compressing objects: 100% (281/281), done.
remote: Total 423 (delta 230), reused 316 (delta 135), pack-reused 0
Receiving objects: 100% (423/423), 604.15 KiB | 727.00 KiB/s, done.
Resolving deltas: 100% (230/230), done.
Submodule 'components/esp-idf-ssd1306' (git@github.com:bkeevil/esp-idf-ssd1306.git) registered for path 'components/esp-idf-ssd1306'
Submodule 'components/esp32-camera' (git@github.com:espressif/esp32-camera.git) registered for path 'components/esp32-camera'
Cloning into '/home/g8ecj/esp/esp32-cam/components/esp-idf-ssd1306'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:bkeevil/esp-idf-ssd1306.git' into submodule path '/home/g8ecj/esp/esp32-cam/components/esp-idf-ssd1306' failed
Failed to clone 'components/esp-idf-ssd1306'. Retry scheduled
Cloning into '/home/g8ecj/esp/esp32-cam/components/esp32-camera'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:espressif/esp32-camera.git' into submodule path '/home/g8ecj/esp/esp32-cam/components/esp32-camera' failed
Failed to clone 'components/esp32-camera'. Retry scheduled
Cloning into '/home/g8ecj/esp/esp32-cam/components/esp32-camera'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:espressif/esp32-camera.git' into submodule path '/home/g8ecj/esp/esp32-cam/components/esp32-camera' failed
Failed to clone 'components/esp32-camera' a second time, aborting
Cloning into '/home/g8ecj/esp/esp32-cam/components/esp-idf-ssd1306'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:bkeevil/esp-idf-ssd1306.git' into submodule path '/home/g8ecj/esp/esp32-cam/components/esp-idf-ssd1306' failed
Failed to clone 'components/esp-idf-ssd1306' a second time, aborting

```

and copy all the files from esp32-cam/components/esp-idf-ssd1306/Main up a level - error was:
```
CMake Error at /home/g8ecj/esp/esp-idf/tools/cmake/component.cmake:468 (add_library):
  Cannot find source file:

    /home/g8ecj/esp/esp32-cam/components/esp-idf-ssd1306/ssd1306.c
```

and after that I get the following error:

```
/home/g8ecj/esp/esp32-cam/components/esp32-camera/driver/twi.c: In function 'pinMode':
/home/g8ecj/esp/esp32-cam/components/esp32-camera/driver/twi.c:61:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc
/home/g8ecj/esp/esp32-cam/components/esp32-camera/driver/twi.c:61:24: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [/home/g8ecj/esp/esp-idf/make/component_wrapper.mk:292: driver/twi.o] Error 1
make: *** [/home/g8ecj/esp/esp-idf/make/project.mk:604: component-esp32-camera-build] Error 2
```
Let me know if my bumbling can assist further!

bkeevil, 2020-01-25:
I should not have assumed that people would have SSH set up with github. My bad. Switching to https mode then. I'm new to git submodules obviously.

bkeevil, 2020-01-25:
Also, with the latest version 3 ESP-IDF some header files were moved. The fix is to change lines 32 and 33 of components/esp32-camera/driver/camera.c to

```
#include "nvs_flash/include/nvs_flash.h"
#include "nvs_flash/include/nvs.h"
```
I just got a clean build with this change

g8ecj, 2020-01-25:
Stepping back a bit I realise that I don't even know which version of the IDF I should be using. I checkout out master (the default) to start with but I think that somehow I got v3.3.1 which I think is current stable? Neither is working for me and I'm desperate to avoid using any Arduino IDE based code (which I hate with a passion!) which is why your project looked to fulfill my requirements.

bkeevil, 2020-01-25:
I wrote it for version 3.x about 6 months ago but it looks like there are breaking changes even in the latest 3.3.1. ESP-IDF is not really done yet. Maybe with version 4 we'll start seeing some API stability. I will work some more on getting the code up to date over the weekend. Be nice to get this project to a point where it can be a demo in espressif's library. More maintainers that way.

bkeevil, 2020-01-25:
I guess I wrote the code for branch release/v4.0 because that seems to be the only version it compiles with. I had to disable SNTP temorarily to get it to compile but the feature was not being used because I haven't completely implemented the font library yet. It is only there to for overlaying dates timestamps on the images.

In any case I got a clean build

g8ecj, 2020-02-03:
Been sidetracked for a week but back onto it! Restarted clean with release/4.0 of idf, checkout out your repo (clean now, no permission errors), build, flashed and running.
Only issue I've found is that if I put in a nonsense ssid, it does the 5 tries but then failed to fire up the AP.
Well done, great starting point for me now ;)

bkeevil, 2020-02-03:
Good to hear. Maybe open another issue on that AP one. I'll close this issue for now.

closed 2020-02-03 by bkeevil