bkeevil/esp32-cam #4: Not really an issue but request to extend documentation

melvinisken, 2020-01-28:
Hi Bond,
thanks for your cool firmware modifications! Now the camera is useful for webcam/smart home usage! (The original examples and most projects I found especially were missing the permanent parameter saving option - great!)
If someone wants to integrate the camera into smart home systems like OpenHAB or ioBroker, it may be usefull to mention the URLs (you can find them out by digging through the system, but maybe not everyone is able to):
Picture: http://IP/capture
Stream: http://IP:81/stream
Control LED: http://IP/control?var=led_intensity&val=[0...255]
Then you can use these urls to create controls in the smart home systems easily. So I thought it makes sense to include them in the description.

melvinisken, 2020-01-28:
And maybe additionally:
I never used the esp toolchain before (only Arduino IDE), and I only had a Windows 10 machine at my desk, I installed the esp toolchain with the standard installer, which ended up in the requirement to use the python commands instead of make (make didn't work at all):

```
idf.py build
idf.py menuconfig
idf.py app
idf.py p- Portname app-flash
```

bkeevil, 2020-01-29:
Oh good I'm glad someone got it working. I may be able to get to this at some point but not right now. Can you create a pull request?

bkeevil, 2020-02-03:
Completed with pull request d75e16b. Thank you for your contributions :-) I will close now.

closed 2020-02-03 by bkeevil

melvinisken, 2020-02-03:
Thanks very much for creating the pull request. I was quite busy the last days and didn't make it. Next time I hope I'm more responsive :-)