bkeevil/esp32-cam #5: Is ONVIF possible?

SargFowler, 2020-01-29:
I'd like to hook up an ESP32-CAM to my NVR recorder, but this requires ONVIF protocol.
Is that something that can be incorporated?

bkeevil, 2020-01-29:
@SargFowler The camera driver only outputs an mjpeg stream and I think ONVIF requires MP4. They work well with the linux motion package if you keep the resolution low.

melvinisken, 2020-01-29:
@SargFowler If you have a spare PC available, it seems that you can turn mjpg-streams into onvif-protocols using an extra SDK: https://www.instructables.com/id/How-to-turn-an-USB-camera-with-Raspberry-Pi-into-a/ (only use step 3, the ozeki sdk). At least it looks like what you need (not a perfect solution though). I expect that you could use a number of mjpeg cameras and convert all of them in parallel with the SDK.

bkeevil, 2020-02-03:
ONVIF 2.0 does support MJPEG streams so it ought to possible to add this feature to the board. It wouldn't be full ONVIF support though. I think I will add it to the project roadmap and maybe someone will take that feature on at some point. Changing to feature request.

SargFowler, 2020-02-03:
That's good to know.
Thanks for the response!

ant1fr1z, 2020-03-15:
onvif really will be very useful when using esp32-sam with aliexpress nvr

redeyedjack, 2020-05-29:
I just got this from reddit for Blue Iris: 
[image]

turcotronics, 2020-05-30:
I confirm the code streams correctly from my ESP32-CAM to a recorder with parameters set to: 
Cam: Generic 
Model: Generic 
Request: http://192.168.0.103:81/stream 
Protocol: Server push (MJPEG) 
Login: empty

Not a standard ONVIF sure but it works :-)
I'll try to play with ONVIF profile S as soon as possible.

SargFowler, 2020-06-04:
Unfortunately, I can't get this to work with my Aliexpress NVR.
I can't set the protocol in the settings, but can confirm the stream appears in a web browser when pointed to the camera address port 81 as above.
I have IPC or ONVIF selection, IP address and port.
Has anyone else been able to connect to their NVR?

ant1fr1z, 2020-06-13:
similarly, aliexpress NVR does not see the flow from esp32cam (((( sad

melvinisken, 2020-07-18:
Maybe it worth to take a look here: https://github.com/enesbcs/ESP32_RTSP_Cam

turcotronics, 2020-07-19:
Thanks, I'll take a look for integration.Rodolfo

DavidAntonin, 2020-09-22:
Hello, is there any news on simple ONVIF implementation? 
Thank you

flatsiedatsie, 2020-09-23:
I don't think anyone has implemented it yet. But from what I [read here](https://forum.arduino.cc/index.php?topic=643396.0) it should be a matter of generating some XML files.

wyne, 2020-11-29:
For anyone else reading this looking for a workaround, this worked very easily for me with Synology Surveillance Station:
https://github.com/circuitrocks/ESP32-RTSP

DavidAntonin, 2020-12-01:
It is not possible to manually specify the RTSP stream in my NVR

DavidAntonin, 2021-03-04:
> I confirm the code streams correctly from my ESP32-CAM to a recorder with parameters set to:
> Cam: Generic
> Model: Generic
> Request: http://192.168.0.103:81/stream
> Protocol: Server push (MJPEG)
> Login: [...]

Any update on Onvif? Thank you

Adorfer, 2021-07-12:
Any news here? having some of those AliExpress NVRs too, who autodetect ONVIF-devices on the network. no possibility o enter the rtmp-address manually, they get it via a web-page from the device.

rtek1000, 2022-01-18:
Hello,

I believe that the idea of using an NVR is to use the P2P App for Internet access,

With that in mind, I found this other platform, which, despite not solving the current case, because it is still not recommended to use a public network,

But it might be interesting to know a little about how the P2P system works:

https://github.com/nabto/nabto-esp32cam

It may be possible in the future to make a secure P2P connection similar to the NVR, using ESP32,

But I believe it may be necessary to use more RAM, the IP cameras I've seen have RAM chips similar to the one used in Linux routers that support OpenWRT OS:

https://openwrt.org/supported_devices

rtek1000, 2022-01-18:
Hi,

I found this option:

Using ngrok tunnel service
After signing up for the account, go to your email address and activate your account by clicking on the activation link.
Now login to your account and click on the ‘Auth’ button’ to get your Tunnel Authtoken as shown in the figure below. Copy this Tunnel Authtoken, we will need it later on.

https://www.elementzonline.com/blog/Accessing-ESP32-CAM-Video-Streaming-from-anywhere-in-the-world

Adorfer, 2022-01-19:
> I believe that the idea of using an NVR is to use the P2P App for Internet access,

never thought of that, but possible that there are actual people with this in mind. 
Or to put it in a different way: 

no, i would need ONVIF to have it on an NVR. on a site without internet access (not enought upstream for 10+ cameras).

rtek1000, 2022-01-19:
If it's to have on a site without internet then just these software wouldn't solve it? I've used the Raspberry Pi with MotionEyeOs, and that was enough for me.

https://www.ispyconnect.com/camera/esp

https://github.com/espressif/esp32-camera/issues/217

> iSpy is described as 'uses your USB webcams, IP cams, capture cards, desktops and microphones to detect and record movement or sound and provides security, surveillance, monitoring and alerting services. Media is recorded directly to H264 mp4 files or AVI files' and is an app in the Video & Movies category. There are more than 50 alternatives to iSpy for a variety of platforms, including Windows, Android, iPhone, Linux and iPad. The best alternative is ZoneMinder, which is both free and Open Source. Other great apps like iSpy are Shinobi (Free Personal, Open Source), AtHome Video Streamer (Free), motionEyeOS (Free, Open Source) and Haven (Free, Open Source).

https://alternativeto.net/software/ispy/

Adorfer, 2022-01-19:
NB: This issue is called "ONVIF", not "how to use Raspi with esp32-cam". Please stay on topic!

rtek1000, 2022-01-19:
Does the ESP32 have enough memory to make an ONVIF server?

rtek1000, 2022-01-19:
https://github.com/onvif/specs/discussions/182

tensiondriven, 2022-05-14:
Note that has a supported build of Tasmota for ESP32 / ESPCAM, and it provides HTTP and RTSP out of the box.  It can also change resolution using the inbuilt console, and has MQTT enabled.  No Onvif though.

ifdreams, 2022-08-30:
This guy has implemented ONVIF on ESP32    [https://www.bilibili.com/video/BV1jg411D7DE?spm_id_from=333.337.search-card.all.click](url)

ant1fr1z, 2022-09-25:
> This guy has implemented ONVIF on ESP32 https://www.bilibili.com/video/BV1jg411D7DE?spm_id_from=333.337.search-card.all.click https://www.youtube.com/watch?v=Oa11X59BZb0&t=4s
looks like a fake, there is no code and the author's page with viruses

icyleaf, 2023-01-29:
> This guy has implemented ONVIF on ESP32 https://www.bilibili.com/video/BV1jg411D7DE?spm_id_from=333.337.search-card.all.click https://www.youtube.com/watch?v=Oa11X59BZb0&t=4s

The author released firmware on github https://github.com/trygoto/esp32-onvif today. I'm not test yet.

codedogz777, 2023-07-10:
Is it possible someone made an English tutorial? I tried and failed doing it

voir64, 2024-12-29:
bonjour.
J ai essayé de récupérer le code  https://github.com/trygoto/esp32-onvif . j arrive à charger  mais à la fin je n 'ai pas rex ni l'ip
ssid : 
ip:
quelqu 'un a réussi ?