bkeevil/esp32-cam #6: error: unknown type name 'nvs_handle_t'

pietropizz, 2020-02-02:
I compile the project on a Mac and I did the set up of the development environment accordingly to your instructions as described in the read.me. The "hello world" app is running without errors.
After "make app" of the esp-32 cam app I get this errors:

CC build/main/app_fonts.o
/Users/gerhardploier/esp/myprojects/esp32-cam/main/app_fonts.c: In function 'app_fonts_main':
/Users/gerhardploier/esp/myprojects/esp32-cam/main/app_fonts.c:351:3: error: unknown type name 'nvs_handle_t'
   nvs_handle_t handle;
   ^
In file included from /Users/gerhardploier/esp/esp-idf/components/nvs_flash/include/nvs.h:20:0,
                 from /Users/gerhardploier/esp/esp-idf/components/nvs_flash/include/nvs_flash.h:21,
                 from /Users/gerhardploier/esp/myprojects/esp32-cam/main/app_fonts.c:6:
/Users/gerhardploier/esp/myprojects/esp32-cam/main/app_fonts.c:357:93: error: pointer targets in passing argument 4 of 'nvs_open_from_partition' differ in signedness [-Werror=pointer-sign]
   ESP_ERROR_CHECK(nvs_open_from_partition(STORAGE_PARTITION,STORAGE_NAMESPACE,NVS_READWRITE,&handle));
                                                                                             ^
/Users/gerhardploier/esp/esp-idf/components/esp32/include/esp_err.h:117:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from /Users/gerhardploier/esp/esp-idf/components/nvs_flash/include/nvs_flash.h:21:0,
                 from /Users/gerhardploier/esp/myprojects/esp32-cam/main/app_fonts.c:6:
/Users/gerhardploier/esp/esp-idf/components/nvs_flash/include/nvs.h:137:11: note: expected 'nvs_handle * {aka unsigned int *}' but argument is of type 'int *'
 esp_err_t nvs_open_from_partition(const char *part_name, const char* name, nvs_open_mode open_mode, nvs_handle *out_handle);

bkeevil, 2020-02-02:
That issue was recently fixed. Pull the latest commit for this repo and the esp32-camera sub module repo.

bkeevil, 2020-02-03:
See also Issue #3

pietropizz, 2020-02-06:
I got it working on the mac! The problem was, that I followed the link posted in your readme file to set up the esp-idf environment. The posted link links you to the stable version  v3.3.1 of ESP-IDF, but you ask to use v4.0 of ESP-IDF. Probably you update the link in your read me file.

You did a great job, it works perfect!

closed 2020-02-06 by pietropizz