bkeevil/esp32-cam #7: Availability of bin files

jd710313, 2020-02-03:
Hi Bond,
This project is really interesting I would love to deploy this to my ESP32 Cam board.  I am in the process of trying to build and flash the project to my board, but find the toolchain and development framework on Windows not so easy to do.
Would it be possible to make the compiled binary (bin) files available so that they could just be flashed to the board?
Thanks
Johan

bkeevil, 2020-02-03:
Apparently you need to use idf.py rather than make on Windows. I updated the README.md file a few hours ago with some info provided by another user.

melvinisken, 2020-02-03:
Since some friends are also interested in this setup, I planned to to a simple step by step instruction, but I didn't find the time so far. 
I can provice such a binary I guess, but it might not help that much since I would have to leave the wifi settings empty / dummy and it was mentioned that the AP functionality could be broken (https://github.com/bkeevil/esp32-cam/issues/3#issuecomment-581265940_) and you cannot login afterwards.

bkeevil, 2020-02-03:
Users need to run `make menuconfig` and choose their board type and specify their settings.  Those settings modify the compiled binary. Even with the ability to store wifi settings in NVRAM there there could potentially be dozens of different permutations of settings to maintain binaries for.

My vote is to improve the instructions for Windows and Mac for now. I don't have either of those platforms so I won't be much help there.

jd710313, 2020-02-04:
Hi Bond, thanks for the explanation and I understand the challenge with multiple configuration permutations and creating bin files.  I will try and make some screenshots if I can get the Windows toolchain and IDF process working.

karlitos, 2020-05-07:
I think offering pre-compiled binaries for some of the most popular boards (mostly AI Thinker ESP32-CAM and all the 5$ clones) will for sure attract more people toward this project. Setting with LED illuminator support OV2640 cam and without LCD support would satisfy 75% of potential users.

The default setting could be without any WiFi Client settings, offering a pre-defines WiFi AP Settings, so the user could connect with the camera in AP mode and just edit his WiFi settings in the GUI.

Other really cool feature making this project more user-friendly would be, if we can support updating the firmware from the SD card. Maybe [this example](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/SD_Update/SD_Update.ino) could provide some starting point.

karlitos, 2020-05-08:
If I am not mistaken, the configuration, done in the _make menuconfig_ step, is then saved in the _sdkconfig_ file in the root directory. So it should be possible to create some default configuration(s) for the ESP32-CAM and by renaming it to _sdkconfig_ skipping the _make menuconfig_ step. With this, automatic builds of multiple .bin files with different setup seems to be also possible with some script. I get a fresh new ESP32-CAM clone by tomorrow, so I try to build such firmware and then flash it. For Windows, there is a [flash-tool with GUI ](http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/), for Mac, there is the [esptool CLI flasher](https://michael-kuehnel.de/iot/2019/07/16/how-to-flash-esp8266-and-esp32-to-use-espruino-firmware-on-mac-os.html). The question is which parameters should be used, the answer lies in the __build__ directory in the _flash\_...\_args_ files

bkeevil, 2020-05-08:
I haven't really looked at flashing a .bin file in much depth but: You will have to read up on the ESP32 partition schemes and look at the partions.csv file. The gzipped copies of the js, html and css files are stored on a separate partition than the bin. Also, there is a partially complete font library that I began working on that will store font bitmaps in a partition. (I was porting a subset of the Adafruit graphics library so I could do text overlays on images but that feature is not finished). So its not quite as simple as just flashing the BIN file, you have to pass all this partition info and extra files to the flashing app as well.

Default settings can go in sdkconfig.defaults but all that stuff is part of the cmake build system that creates the BIN file for flashing.

I think this issue is a can of worms TBH. A user friendly build system is Espressif's problem not ours.

closed 2021-02-11 by bkeevil