bkeevil/esp32-cam #11: ESP32-cam doesnt connect to local wifi in station mode but only as AP

Ciro67, 2020-03-15:
I've build and downloaded the SW.
At start up after several attempts the ESP32-Cam is configured as AP.
I've tried to change password of my Home ap but the behavior is always the some.
Some times happen also that the reboot is continuously and need to power off and power on .
Below the Serial Monitor output 

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[0;32mI (277) psram: This chip is ESP32-D0WD[0m
[0;32mI (277) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (277) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (280) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (287) cpu_start: Pro cpu up.[0m
[0;32mI (291) cpu_start: Application information:[0m
[0;32mI (296) cpu_start: Project name:     esp32-cam[0m
[0;32mI (301) cpu_start: App version:      9bb7a8f[0m
[0;32mI (306) cpu_start: Compile time:     Mar 15 2020 00:36:32[0m
[0;32mI (312) cpu_start: ELF file SHA256:  14ca017eb2434a72...[0m
[0;32mI (318) cpu_start: ESP-IDF:          v4.0-dirty[0m
[0;32mI (323) cpu_start: Starting app cpu, entry point is 0x400815f0[0m
[0;32mI (315) cpu_start: App cpu up.[0m
[0;32mI (824) spiram: SPI SRAM memory test OK[0m
[0;32mI (824) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (824) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (831) heap_init: At 3FFBC600 len 00023A00 (142 KiB): DRAM[0m
[0;32mI (837) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (843) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (850) heap_init: At 4009CF20 len 000030E0 (12 KiB): IRAM[0m
[0;32mI (856) cpu_start: Pro cpu start user code[0m
[0;32mI (861) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (881) spi_flash: detected chip: generic[0m
[0;32mI (881) spi_flash: flash io: qio[0m
[0;32mI (881) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (949) settings: NVS Flash Init[0m
[0;32mI (949) settings: Settings loaded from NVS[0m
[0;32mI (949) settings:  hostname=espressif[0m
[0;32mI (949) settings:  wifi_ssid=SDCfamily[0m
[0;32mI (959) settings:  wifi_password=xxxxxxxxxxxxxxxxxxxxpgerico[0m
[0;32mI (959) settings:  mdns_instance=ESP32 Camera Web Server[0m
[0;32mI (969) settings:  ntp_server=pool.ntp.org[0m
[0;32mI (969) settings:  timezone=[0m
[0;32mI (979) settings:  dhcp=1[0m
[0;32mI (979) settings:  ip=0.0.0.0[0m
[0;32mI (979) settings:  netmask=0.0.0.0[0m
[0;32mI (989) settings:  gateway=0.0.0.0[0m
[0;32mI (989) settings:  dns1=0.0.0.0[0m
[0;32mI (989) settings:  dns2=0.0.0.0[0m
[0;32mI (999) sccb: pin_sda 26 pin_scl 27
[0m
[0;32mI (999) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1099) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1099) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1109) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1119) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1129) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1139) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1149) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1159) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1169) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1179) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1189) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (1189) camera: Allocating 2 frame buffers (117 KB total)[0m
[0;32mI (1199) camera: Allocating 58 KB frame buffer in OnBoard RAM[0m
[0;32mI (1209) camera: Allocating 58 KB frame buffer in OnBoard RAM[0m
I (1459) wifi: wifi driver task: 3ffda228, prio:23, stack:3584, core=0
[0;32mI (1459) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (1459) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (1479) wifi: wifi firmware version: 581f422
I (1479) wifi: config NVS flash: enabled
I (1479) wifi: config nano formating: disabled
I (1479) wifi: Init dynamic tx buffer num: 32
I (1489) wifi: Init data frame dynamic rx buffer num: 32
I (1489) wifi: Init management frame dynamic rx buffer num: 32
I (1499) wifi: Init management short buffer num: 32
I (1499) wifi: Init static rx buffer size: 1600
I (1509) wifi: Init static rx buffer num: 10
I (1509) wifi: Init dynamic rx buffer num: 32
[0;32mI (1519) wifi station: Connecting to AP SSID:SDCfamily password:1q2w3e4r5t6y7u8i9o0pgerico[0m
I (1519) wifi: set country: cc=

bkeevil, 2020-03-15:
Did you set the SSID and password through `make menuconfig` or did you store it in nvram through the web interface?

bkeevil, 2020-03-15:
Also, what are you using for a power supply? Wifi requires quite a bit of current and if it doesn't get enough the ESP32 browns out and reboots. A USB connection from a computer usually isn't enough.

Ciro67, 2020-03-15:
Hi Thanks for your immediate response.
Ive configured SSID and PWD by menuconfig.
The power is coming from usb 
But the esp work if i use arduino cam example, without any issue.
at moment I'm closed in my house ( im in Italy) and cannot going in my laboratory to take the additional power.
But the board  work in AP mode and i can connect to it.

Ciro67, 2020-03-15:
ok
solved thanks.
was a power issue.
Ive used phone charger and now work.
thanks so much
Ciro

closed 2020-03-15 by bkeevil