bkeevil/esp32-cam #12: Got some compilation errors during make app

superfino, 2020-03-25:
Hi, i followed exactly your describtion but now i'm struggling with make app.  I used esp-idf v4.2-dev-792-g6330b3345 i think the actual version. Everything worked fine until these compiler errors came up. Tried to find out what went wrong but the messages are not that much precise for me.
Thanks for any advice..  Jo
[comp_errors.txt](https://github.com/bkeevil/esp32-cam/files/4383465/comp_errors.txt)

bkeevil, 2020-03-25:
It only seems to compile with IDF 4.0

superfino, 2020-03-26:
You're right. Went back to IDF 4.0 and everything worked like a charm. Thanks for your fast response and this ESP32-cam piece of code. The best I've seen so far.

closed 2020-03-26 by superfino