bkeevil/esp32-cam #14: Different name of the CSS stylesheet referenced in the index.html

karlitos, 2020-05-06:
In the __index.html __ page, the stylesheet is referenced as __stylesheet.css__
```
<link rel="stylesheet" type="text/css" href="stylesheet.css" media="screen" />
```
but the name of the stylesheet is __style.css__. This is confusing and also breaks the support of the IDEs. Renaming the stylesheet reference to __style.css__ results in unstyled webpage.

So, somewhere along the process of building the app, the stylesheet is renamed to __stylesheet.css__
1. Is there a reason for this ?
2. Is this really necessary ?

karlitos, 2020-05-06:
It turned out the "fix" can be quite easy. In the `app_httpd.c` change the `httpd_uri_t stylesheet_uri` to:

```c
 httpd_uri_t stylesheet_uri = {
        .uri       = "/style.css",
        .method    = HTTP_GET,
        .handler   = stylesheet_handler,
        .user_ctx  = NULL
    };
```

For someone, who does Web-Development as me it is really confusing when the HMTL references non-existing stylesheet.

bkeevil, 2020-05-06:
This was inherited from the original ESP-WHO demo app that the project was forked from. I am not aware of any reasons why it was done this way. Go ahead and fix.

bkeevil, 2020-05-06:
Merged. Thank you for your contribution karlitos

closed 2020-05-06 by bkeevil