bkeevil/esp32-cam #15: The LED illuminator range set in the settings is not reflected in the HTML

karlitos, 2020-05-06:
Even thought if I set the maximum level of the LED illuminator under under _Camera Web Server_ -> _LED Illuminator_ the slider-range for the illuminator is har-coded to go from 0 to 255. When the defined maximum range could be retrieved as other settings, it would be easy to adjust the slider range in the Webpage.

karlitos, 2020-05-06:
The fix was easier as thought.

In `app_httpd.c` I extended the __status_handler__:
```c
#ifdef CONFIG_LED_ILLUMINATOR_ENABLED
    p+= sprintf(p, ",\"led_intensity\":%u", led_duty);
    p+= sprintf(p, ",\"led_max_intensity\":%u", CONFIG_LED_MAX_INTENSITY);
```

And in the `script.js` javascript file I extended the `fetchSettings`:
```javascript
if (state.led_intensity !== -1) {
          let led_intensity_slider = document.getElementById("led_intensity");
          led_intensity_slider.max = state.led_max_intensity;
          let led_intensity_range_max = document.querySelector('#led-group > div.range-max');
          led_intensity_range_max.innerText = state.led_max_intensity;
        }
```
I try to do a propper PR

bkeevil, 2020-05-06:
Can you put the stylesheet rename in the PR too?

karlitos, 2020-05-06:
Sure, I try to find some time soon

bkeevil, 2020-05-06:
If I remember correctly, the original behavior was that the user would adjust led_max_intensity and then they would have to "save settings" before the led_intensity slider would adjust to the new range.

closed 2020-05-07 by bkeevil