bkeevil/esp32-cam #16: Support for onboard LEDs - not the flash/illuminator

karlitos, 2020-05-06:
Most people probably have some of those cheap copies of the _Ai Thinker ESP32-CAM_ modules. Those have a onboard (red) LED. It would be great, if the LED could be configured to display the current board status.

E.g.:

0. Always off
1. On when board running
2. On when connected to WiFi
3. Fast blinking when connecting to WiFi
4. Slow blinking when in AP mode

bkeevil, 2020-05-06:
I would suggest implementing this as a FreeRTOS task and not using an LEDC channel. Also, since not all boards have an onboard LED and may use different PIN numbers, this extension should need to be enabled and configured via MAKE MENUCONFIG. 

See https://github.com/bkeevil/esp32-cam/blob/master/main/Kconfig.projbuild to add settings

karlitos, 2020-05-06:
Yes, I was thinking about making a new menu entry in the MAKE MENUCONFIG step. Something like:
- [x] Enable support for onboard LED
  - [ ] On when board running
  - [ ] On when connected to WiFi
  - [x] Fast blinking when connecting to WiFi
  - [ ] Slow blinking when in AP mode
  - LED Pin [ 1 ]
 
But I have no experience with FreeRTOS. Coming from the Arudino world I was thinking about blinking the LED in the `app_main()` 😬

bkeevil, 2020-05-06:
Yeah for this feature you would need to learn about FreeRTOS EventGroups and Tasks. The Wifi library throws events that must be caught by your task. app_lcd.c is a good example of adding a task.

karlitos, 2020-05-08:
Thank you for inviting me to this project. I will try to take care of this enhancement, I will open a branch for it and will post my changes to it, but for sure I will need some "mentoring" and consultations. But utilizing the onbaord in a FreeRTOS task LED seems to be doable.

bkeevil, 2020-05-08:
Yeah why not create your own branch just in case something goes horribly wrong. Let me know if you run into any permissions issues.