bkeevil/esp32-cam #17: Support for setting hostname in the MAKE MENUCONFIG

karlitos, 2020-05-06:
Maybe I missed something, but I was not able to find a way to set the hostname in the _make menuconfig_ step. I was able to change the hostname in the Web-UI, the setting is getting saved and also it's reflected in the header.

But my cam is still known as __espressif__ in my home network, even after several reboots. Is it a network issue or some bug related to the firmware ?

bkeevil, 2020-05-06:
I think you just need to add a call to `tcp_adapter_set_hostname()` in `app_wifi.c` before line 198: `ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(127));`

Maybe see if that works. I don't have a working development environment to test anything atm.

karlitos, 2020-05-07:
I see two `tcp_adapter_set_hostname()` calls just at the beginning of `app_wifi_startup()`. I created a PR #22 adding the menu entry - since I believe it makes sense for the user to set the hostname prior to flash. I will test whether this will fix my issue, or not.

bkeevil, 2020-05-07:
They may be in the wrong place?

karlitos, 2020-05-07:
I think this was a network issue. After flashing an image with the hostname set-up in the make menuconfig, the cam still appeared in the network as __espressif__ but now, suddenly, it changed and I can reach the cam under the right hostname. You may close this issue, since the actual enhancement was already implemented

closed 2020-05-07 by bkeevil

melvinisken, 2020-05-08:
Just a comment: within my setup, the hostname is in the sdkconfig file (I think it will be filled after the first run of make menuconfig, but maybe with standard settings and without the ability to edit):
CONFIG_LWIP_LOCAL_HOSTNAME=""
The string given here is also reflected as local hostname in my network...

karlitos, 2020-05-08:
Yes, that's true, this is the config property for the hostname and in my PR I introduced option in _make menuconfig_ allowing to set it before the compilation. Again, I don't know why I could not reach my camera under "my" hostname, even I set the hostname in the Web-UI, but now it's fixed. Either it was a network issue - like the router keeps the hostnames for some time or the new config option fixed it. What should be tested: if it's possible to change the hostname in the Web-UI and then reach the camera under the new hostname.