bkeevil/esp32-cam #24: Can't configure a SSID in spaces with the Network Settings

superfino, 2020-05-08:
Unfortunately I can't change my Network Settings in the menu. Tried to set a SSID with a space charakter doesn't work. The space character becomes replaced by %20 what I can see when I connect to the AP. Is there anything special when using a space character in a SSID. Any help will be appreciated.

bkeevil, 2020-05-08:
I never tested SSIDs with space characters. We need to convert html special characters back to the original string near line 299 of app_httpd.c

Is there an existing ESP-IDF function to do this? I can't seem to find one in the http_parser.h file

bkeevil, 2020-05-08:
From stackoverflow:

```
Section 7.3.2.1 of the 802.11-2007 specification (http://standards.ieee.org/getieee802/download/802.11-2007.pdf) defines SSIDs.

A valid SSID is 0-32 octets with arbitrary contents. A 0-length SSID indicates the wildcard SSID (in probe request frames for instance).

There's no character set associated with the SSID - a 32-byte string of NUL-bytes is a valid SSID.

This implies:

    - you should never use normal string functions when manipulating generic SSIDs (strcpy() and friends).

    - you should not assume that the SSID is printable when, for instance, logging it to disk
```

superfino, 2020-05-08:
Thank you for your clarification. It sounds that there is actually no solution or circumvention when i understand you right.

bkeevil, 2020-05-09:
I think this one needs to be fixed.