bkeevil/esp32-cam #26: esp-32 does not connect to WiFi in STA mode neither switches to AP mode

lstankiewicz, 2020-07-23:
Hello,

I finally managed to compile the firmware using esp-idf v4.0.1 and flash it to the AI Thinker board.
I connected the debug serial and I saw that when connecting to WiFi in STA mode it doesn't show neither SSID nor pass.
I also does not time out.

Here's the output:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7324
load:0x40078000,len:15592
load:0x40080400,len:4976
entry 0x40080704
I (64) boot: Chip Revision: 1
I (65) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.0.1 2nd stage bootloader
I (39) boot: compile time 16:36:13
I (39) boot: Enabling RNG early entropy source...
I (44) qio_mode: Enabling default flash chip QIO
I (49) boot: SPI Speed      : 80MHz
I (53) boot: SPI Mode       : QIO
I (57) boot: SPI Flash Size : 4MB
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 factory          factory app      00 00 00010000 00380000
I (79) boot:  1 fonts            WiFi data        01 02 00390000 00040000
I (87) boot:  2 nvs              WiFi data        01 02 003d0000 00004000
I (94) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24d40 (150848) map
I (155) esp_image: segment 1: paddr=0x00034d68 vaddr=0x3ffb0000 size=0x03b40 ( 15168) load
I (160) esp_image: segment 2: paddr=0x000388b0 vaddr=0x40080000 size=0x00400 (  1024) load
I (162) esp_image: segment 3: paddr=0x00038cb8 vaddr=0x40080400 size=0x07358 ( 29528) load
I (181) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x942bc (606908) map
I (344) esp_image: segment 5: paddr=0x000d42dc vaddr=0x40087758 size=0x15c70 ( 89200) load
I (391) boot: Loaded app from partition at offset 0x10000
I (391) boot: Disabling RNG early entropy source...
I (392) psram: This chip is ESP32-D0WD
I (396) spiram: Found 64MBit SPI RAM device
I (401) spiram: SPI RAM mode: flash 80m sram 80m
I (406) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (413) cpu_start: Pro cpu up.
I (417) cpu_start: Application information:
I (422) cpu_start: Project name:     esp32-cam
I (427) cpu_start: App version:      a71149d
I (432) cpu_start: Compile time:     Jul 23 2020 16:31:58
I (438) cpu_start: ELF file SHA256:  073d4299a4ceba37...
I (444) cpu_start: ESP-IDF:          v4.0.1
I (449) cpu_start: Starting app cpu, entry point is 0x400817a0
I (0) cpu_start: App cpu up.
I (949) spiram: SPI SRAM memory test OK
I (950) heap_init: Initializing. RAM available for dynamic allocation:
I (950) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (956) heap_init: At 3FFBCE70 len 00023190 (140 KiB): DRAM
I (962) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (968) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (975) heap_init: At 4009D3C8 len 00002C38 (11 KiB): IRAM
I (981) cpu_start: Pro cpu start user code
I (986) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1006) spi_flash: detected chip: generic
I (1006) spi_flash: flash io: qio
I (1006) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1074) settings: NVS Flash Init
I (1074) settings: Settings loaded from NVS
[... 776 lines cut ...]
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options
`
Am I doing sth wrong or missing some additional config or is it a bug?

Thank you for your help

melvinisken, 2020-07-24:
Attached see my working config. Just do a file compare to see the differences. I did not check too closely, but it looks like you didn't follow the "important settings" section in the readme. At least you didn't pin WIFI to core 0. There are some other differences as well (but I'm not using the latest version I think, so not all differences might be relevant).
[sdkconfig.txt](https://github.com/bkeevil/esp32-cam/files/4972396/sdkconfig.txt)

bkeevil, 2020-07-24:
Yeah the code needs to be updated to the latest version of ESP-IDF

I don't think it likes the fact that your AP is on channel 14. Channel 14 is not allowed in much of the world so there could be an issue with your country settings. Can you try a different channel on your router?

> I (1594) wifi:set country: cc= schan=1 nchan=14 policy=0
> W (1604) wifi:External AP's channel is out of channel range, disconnect

My boards ignored country settings and tx power settings anyways so there may not be much point in playing with those.

lstankiewicz, 2020-07-25:
@melvinisken 
In my config :
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
so it is pinned to CORE_0 but in your config it is pinned to CORE_1:
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1=y

I have also tried your sdkconfig.txt but the result was exactly the same.

@bkeevil I have set my phone to act as an AP, just to exclude poor WiFi range and unfortunately Android's tethering doesn't let me play with channels.
I also couldn't find those settings in the menuconfig.

I suspected that menuconfig does't store my settings, but I have checked in build/config/sdkconfig.h and all was there.
I also tried:
`make clean`
`rm sdkconfig*`
`make menuconfig `
(configure all from scratch)
`make app`
`make flash`

Result exactly the same

Next I enabled WiFi debugging and this is what I got:

`I (1468) settings: NVS Flash Init`
`I (1468) settings: Settings loaded from NVS`
`I (1468) settings:  hostname=espressif`
`I (1478) settings:  wifi_ssid=`
`I (1478) settings:  wifi_password=`
`I (1478) settings:  mdns_instance=ESP32 Camera Web Server`
`I (1488) settings:  ntp_server=pool.ntp.org`
`I (1488) settings:  timezone=`
`I (1498) settings:  dhcp=1`
`I (1498) settings:  ip=0.0.0.0`
`I (1498) settings:  netmask=0.0.0.0`
`I (1508) settings:  gateway=0.0.0.0`
`I (1508) settings:  dns1=0.0.0.0`
`I (1518) settings:  dns2=0.0.0.0`
`I (1518) sccb: pin_sda 26 pin_scl 27`
``
`I (1528) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0`
`I (1628) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1628) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1638) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1648) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1658) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1668) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1678) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1688) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1698) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1698) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1708) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0`
`I (1718) camera: Allocating 2 frame buffers (117 KB total)`
`I (1738) camera: Allocating 58 KB frame buffer in OnBoard RAM`
`I (1748) camera: Allocating 58 KB frame buffer in OnBoard RAM`
`I (2008) wifi:wifi driver task: 3ffd4bfc, prio:23, stack:6656, core=0`
`I (2008) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE`
`I (2008) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE`
`I (2028) wifi:wifi firmware version: aa5336b`
`I (2028) wifi:config NVS flash: enabled`
`I (2028) wifi:config nano formating: disabled`
`I (2028) wifi:Init dynamic tx buffer num: 32`
`I (2038) wifi:Init data frame dynamic rx buffer num: 32`
`I (2038) wifi:Init management frame dynamic rx buffer num: 32`
`I (2048) wifi:Init management short buffer num: 32`
`I (2048) wifi:Init static tx buffer num: 16`
`I (2058) wifi:Init static rx buffer size: 1600`
`I (2058) wifi:Init static rx buffer num: 10`
`I (2068) wifi:Init dynamic rx buffer num: 32`
`I (2068) wifi station: Connecting to AP SSID: password:`
`D (2078) wifi:clear blacklist`
`I (2078) wifi:set country: cc= schan=1 nchan=14 policy=0`
``
`W (2078) wifi:External AP's channel is out of channel range, disconnect`
`I (2088) wifi station: wifi_init_sta finished.`
`I (2088) wifi station: connect to ap SSID: password:`
`I (2188) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0`
`D (2188) wifi:filter: set rx policy=0`
`I (2188) wifi:mode : sta (3c:71:bf:f8:25:dc)`
`D (2188) wifi:filter: set rx policy=1`
`D (2188) wifi:connect status 0 -> 0`
`D (2198) wifi:Start wifi connect`
`D (2198) wifi:connect status 0 -> 0`
`D (2198) wifi:sta_connect: invalid ssid`
``

An now it looks like it ignores WiFi, NTP and most likely some other settings.
Any idea why?

bkeevil, 2020-07-25:
> W (2078) wifi:External AP's channel is out of channel range, disconnect

On line 121 of app_wifi.c change `.nchan = 14` to `.nchan = 11`, recompile and try again.

Please refer to the function esp_wifi_set_country in https://docs.espressif.com/projects/esp-idf/en/v4.0.1/api-reference/network/esp_wifi.html

bkeevil, 2020-07-25:
Are you in Japan?

It said somewhere the esp_wifi_set_country() should be called after esp_wifi_start() but in my code it is called before. So maybe try moving that call. 

Insert the following code between lines 197 and 198 of app_wifi.c and see if that works.

```
    wifi_country_t country = {
        .cc = "",
        .schan = 1,
        .nchan = 11, 
        .max_tx_power = 20, 
        .policy = WIFI_COUNTRY_POLICY_AUTO,
    };
    ESP_ERROR_CHECK(esp_wifi_set_country(&country));
```

lstankiewicz, 2020-07-26:
@bkeevil 

> Are you in Japan?

I wish :), PL is my home.

I did as you suggested, moved the part of the code and changed parameters,  but that didn't help. 
As I still have WiFi debug enabled i got this:
`[...]`
`I (2069) wifi station: Connecting to AP SSID: password:`
`D (2079) wifi:clear blacklist`
`I (2079) wifi station: wifi_init_sta finished.`
`I (2079) wifi station: connect to ap SSID: password:`
`I (2179) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0`
`D (2179) wifi:filter: set rx policy=0`
`I (2179) wifi:mode : sta (3c:71:bf:f8:25:dc)`
`D (2179) wifi:filter: set rx policy=1`
`D (2179) wifi:connect status 0 -> 0`
`D (2189) wifi:Start wifi connect`
`D (2189) wifi:connect status 0 -> 0`
`D (2189) wifi:sta_connect: invalid ssid`
`I (2199) wifi:set country: cc= schan=1 nchan=11 policy=0`

Interesting part is the line next to last one:
`D (2189) wifi:sta_connect: invalid ssid`

I looked at the code and found in app_wifi_c :
`wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();`
and then 
`ESP_ERROR_CHECK(esp_wifi_init(&cfg));`
So if I read the code correctly it first builds a config `cfg` using esp-idf function WIFI_INIT_CONFIG_DEFAULT() and then initializes WiFi with that config.
My conclusion is that the function WIFI_INIT_CONFIG_DEFAULT() doesn't get the necessary parameters thus the messages:
`I (2069) wifi station: Connecting to AP SSID: password:`
`I (2079) wifi station: connect to ap SSID: password:`
`D (2189) wifi:sta_connect: invalid ssid`

I didn't find any other reference to that function in your code so I assume it is part of esp-idf code.

How can I force to start WiFi in AP mode at start as this would be my go-to setting?

bkeevil, 2020-07-26:
That looks like a second error. In a previous log it looked like it was loading blank values from NVRAM and using them to overwrite the defaults you had stored in the config file. So we have to delete your stored nvram config next.

On line 39 of `app_settings.c` there is a function called `app_settings_reset()` that does this.

Lets call that at line 46 of app_main.c (before the call to `app_settings_startup()`)

bkeevil, 2020-07-26:
I duplicated the error by upgrading to the "latest" version of ESP-IDF 4.0.1

This issue appears relevant: https://github.com/espressif/esp-idf/issues/5013

lstankiewicz, 2020-07-26:
So again I did `make clean` then `make menuconfig`, loaded sdkconfig, saved all then changed/verified the files  `app_wifi.c`  and `app_settings.c`. 
I also edited `Kconfig.buildproj` and set WiFi STA and AP settings.

double-checked files in `build/config` if all settings are there (they were) and 
`make app`
`make flash`

Unfortunately that didn't change a thing :

> ets Jun  8 2016 00:22:57

>rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6960
ho 0 tail 12 room 4
load:0x40078000,len:14756
load:0x40080400,len:4340
entry 0x400806e8
I (69) boot: Chip Revision: 1
I (70) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0.1 2nd stage bootloader
I (41) boot: compile time 23:09:43
I (41) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 80MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 factory          factory app      00 00 00010000 00380000
I (76) boot:  1 fonts            WiFi data        01 02 00390000 00040000
I (84) boot:  2 nvs              WiFi data        01 02 003d0000 00004000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x251b8 (151992) map
I (156) esp_image: segment 1: paddr=0x000351e0 vaddr=0x3ffb0000 size=0x03b80 ( 15232) load
I (162) esp_image: segment 2: paddr=0x00038d68 vaddr=0x40080000 size=0x00400 (  1024) load
I (164) esp_image: segment 3: paddr=0x00039170 vaddr=0x40080400 size=0x06ea0 ( 28320) load
I (182) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x94e00 (609792) map
I (362) esp_image: segment 5: paddr=0x000d4e20 vaddr=0x400872a0 size=0x16458 ( 91224) load
I (413) boot: Loaded app from partition at offset 0x10000
I (413) boot: Disabling RNG early entropy source...
I (413) psram: This chip is ESP32-D0WD
I (418) spiram: Found 64MBit SPI RAM device
I (422) spiram: SPI RAM mode: flash 80m sram 40m
I (428) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (435) cpu_start: Pro cpu up.
I (439) cpu_start: Application information:
I (443) cpu_start: Project name:     esp32-cam
I (449) cpu_start: App version:      a71149d-dirty
I (454) cpu_start: Compile time:     Jul 26 2020 23:04:48
I (460) cpu_start: ELF file SHA256:  4c2615c30c017789...
I (466) cpu_start: ESP-IDF:          v4.0.1
I (471) cpu_start: Starting app cpu, entry point is 0x400817c4
I (0) cpu_start: App cpu up.
I (1362) spiram: SPI SRAM memory test OK
I (1362) heap_init: Initializing. RAM available for dynamic allocation:
I (1363) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1369) heap_init: At 3FFBCF00 len 00023100 (140 KiB): DRAM
I (1375) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1381) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1388) heap_init: At 4009D6F8 len 00002908 (10 KiB): IRAM
I (1394) cpu_start: Pro cpu start user code
I (1399) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1420) spi_flash: detected chip: generic
I (1420) spi_flash: flash io: dio
I (1420) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 36 lines cut ...]
I (2019) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2039) wifi:wifi firmware version: aa5336b
I (2039) wifi:config NVS flash: enabled
I (2039) wifi:config nano formating: disabled
I (2039) wifi:Init dynamic tx buffer num: 32
I (2049) wifi:Init data frame dynamic rx buffer num: 32
I (2049) wifi:Init management frame dynamic rx buffer num: 32
I (2059) wifi:Init management short buffer num: 32
I (2059) wifi:Init static tx buffer num: 16
I (2069) wifi:Init static rx buffer size: 2212
I (2069) wifi:Init static rx buffer num: 10
I (2079) wifi:Init dynamic rx buffer num: 32
I (2079) wifi station: Connecting to AP SSID: password:
D (2089) wifi:clear blacklist
I (2089) wifi station: wifi_init_sta finished.
I (2089) wifi station: connect to ap SSID: password:
I (2189) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
D (2189) wifi:filter: set rx policy=0
I (2189) wifi:mode : sta (3c:71:bf:f8:25:dc)
D (2189) wifi:filter: set rx policy=1
D (2189) wifi:connect status 0 -> 0
D (2199) wifi:Start wifi connect
D (2199) wifi:connect status 0 -> 0
D (2199) wifi:sta_connect: invalid ssid
I (2209) wifi:set country: cc= schan=1 nchan=11 policy=0

bkeevil, 2020-07-26:
Place the calls to app_settings_reset() and app_settings_save() after the call to app_settings_startup()

```
  ESP_ERROR_CHECK(esp_event_loop_create_default());   
  event_group = xEventGroupCreate();
    
  app_settings_startup();    
//  app_settings_reset();
//  app_settings_save();

  app_camera_startup();
  #ifdef CONFIG_LED_ILLUMINATOR_ENABLED
  app_illuminator_startup();
  #endif
  app_wifi_startup();
``
I just got mine working by resetting the nvram settings

bkeevil, 2020-07-26:
I pushed the code to master. Just git pull and uncomment those two lines

lstankiewicz, 2020-07-27:
new day, new challenges 

That did the trick ... to some extent. 
Now the ESP connects to WiFi in STA mode, OLED 128x32 display is working in soft-i2c mode and I can play around ... but when it fails to connect to WiFi it reboots:

>I (1819) wifi station: connecting to ap SSID:myssid password:myp@zw0rd
I (1909) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1909) wifi:mode : sta (3c:71:bf:f8:25:dc)
I (1909) wifi:set country: cc= schan=1 nchan=11 policy=0

>W (3239) wifi station: Station disconnected (reason=201)
I (3239) wifi station: Retry to connect to the AP
W (4559) wifi station: Station disconnected (reason=201)
I (4559) wifi station: Retry to connect to the AP
W (5889) wifi station: Station disconnected (reason=201)
I (5889) wifi station: Retry to connect to the AP
W (7209) wifi station: Station disconnected (reason=201)
I (7219) wifi:flush txq
I (7219) wifi:stop sw txq
I (7219) wifi:lmac stop hw txq
I (7219) wifi station: Restarting wifi in SoftAP mode.
I (7229) wifi station: wifi_init_softap finished.SSID:ESP32CAM password:12345678
I (7239) wifi:mode : softAP (3c:71:bf:f8:25:dd)
I (7239) wifi:Total power save buffer number: 16
I (7239) wifi:Init max length of beacon: 752/752
I (7239) wifi:Init max length of beacon: 752/752
I (7249) camera_httpd: Starting web server on port: '80'
I (7259) camera_httpd: Starting stream server on port: '81'
I (7259) LCD: Starting up LCD Driver
I (7259) LCD: INTERFACE is i2c
I (7269) LCD: CONFIG_SDA_GPIO=15
I (7269) LCD: CONFIG_SCL_GPIO=14
I (7279) LCD: CONFIG_RESET_GPIO=-1
I (7279) LCD: Panel is 128x32
I (7279) SSD1306: OLED configured successfully
ESP_ERROR_CHECK failed: esp_err_t 0x3007 (ESP_ERR_WIFI_CONN) at 0x400977b4
file: "/home/user/esp/myprojects/esp32-cam/main/app_wifi.c" line 51
func: wifi_get_rssi
expression: esp_wifi_sta_get_ap_info(&apinfo)

>ELF file SHA256: b7bd017d8dba7026

>Backtrace: 0x400971d5:0x3fff5740 0x400977b7:0x3fff5760 0x400d7305:0x3fff5780 0x400d7751:0x3fff57f0 0x4009aa71:0x3fff5810

>Rebooting...

It tries to return RSSI but in AP mode it's not necessary, you can get RSSI value for each connected client but not a global value.
I think in AP mode `wifi_get_rssi();` shouldn't be called at all.

mase76, 2021-01-24:
I have a similar problem. When I do not enter a valid ssid and pw in the menuconfig, or just enter "xxx" for both, the device does not enter AP mode:

`I (2214) wifi_init: rx ba win: 6I (2214) wifi_init: rx ba win: 6
I (2214) wifi_init: tcpip mbox: 32
I (2224) wifi_init: udp mbox: 6
I (2224) wifi_init: tcp mbox: 6
I (2224) wifi_init: tcp tx win: 5744
I (2234) wifi_init: tcp rx win: 5744
I (2234) wifi_init: tcp mss: 1440
I (2244) wifi_init: WiFi IRAM OP enabled
I (2244) wifi_init: WiFi RX IRAM OP enabled
I (2254) wifi station: Connecting to AP SSID: password:
I (2254) wifi station: wifi_init_sta finished.
I (2264) wifi station: connecting to ap SSID: password:
I (2354) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (2354) wifi:mode : sta (e8:db:84:12:6e:78)
I (2354) wifi:set country: cc= schan=1 nchan=11 policy=0

I (2214) wifi_init: tcpip mbox: 32
I (2224) wifi_init: udp mbox: 6
I (2224) wifi_init: tcp mbox: 6
I (2224) wifi_init: tcp tx win: 5744
I (2234) wifi_init: tcp rx win: 5744
I (2234) wifi_init: tcp mss: 1440
I (2244) wifi_init: WiFi IRAM OP enabled
I (2244) wifi_init: WiFi RX IRAM OP enabled
I (2254) wifi station: Connecting to AP SSID: password:
I (2254) wifi station: wifi_init_sta finished.
I (2264) wifi station: connecting to ap SSID: password:
I (2354) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (2354) wifi:mode : sta (e8:db:84:12:6e:78)
I (2354) wifi:set country: cc= schan=1 nchan=11 policy=0
`

When configuring a valid Wifi, I also have to uncomment the 2 lines for reset and save.

closed 2021-02-11 by bkeevil