bkeevil/esp32-cam #28: Error during wpa_supplicant.a part of the build process

teixeluis, 2020-08-09:
While trying to build the project from the master branch, and upon having followed the project setup steps in a Linux Debian machine, I have obtained the following error during the "idf.py build" (was able to successfully setup esp-idf prior to that):

```
/home/lteixeira/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(crypto_mbedtls.c.obj): in function `crypto_ec_point_cmp':
/home/lteixeira/projectos/esp32/esp/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:601: undefined reference to `mbedtls_ecp_point_cmp'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Can you please advise on what I may be missing?

The sdkconfig I am using is as follows:

[sdkconfig.txt](https://github.com/bkeevil/esp32-cam/files/5047825/sdkconfig.txt)

One aspect worth mentioning is that I could not find in menuconfig where to configure the serial port to use. In the "serial flasher config" section I only have the following options:

[image]

My build environment OS version is:

```
lteixeira@lteixeira-debian-vm:~/projectos/esp32/esp32-cam$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

bkeevil, 2020-08-09:
That is the error message you get when you use the new version of ESP-IDF. Use the older version referenced in the instructions.

teixeluis, 2020-08-09:
Hi @bkeevil, thanks for the feedback. I have then checked out the release/v4.0 version of esp-idf  (indeed ommited this step in the initial attempt), and then I:

 * executed the `esp-idf install.sh` again;
 * ran the  `. ./export.sh` in order to update the env;
 * under esp32-cam, performed idf.py fullclean, followed by `idf.py menuconfig`, and ran through the configurations to do some double checking;
 * after running `idf.py build` unfortunately I still get the a similar error, albeit pointing to a different line in their lib source code:

```
/home/lteixeira/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(crypto_mbedtls.c.obj): in function `crypto_ec_point_cmp':
/home/lteixeira/projectos/esp32/esp/esp-idf/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:602: undefined reference to `mbedtls_ecp_point_cmp'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
Is there anything else that should be done, i.e. any missing cleanup after switching to  a different version of esp-idf?

Thanks

Best regards

teixeluis, 2020-08-09:
Btw (perhaps overkill), cloned again the esp32-cam repo, started the build again, but the error still occurs.

Cheers

bkeevil, 2020-08-09:
I would delete the entire build directory in the esp32-cam directory and also delete and recreate sdkconfig. If that doesn't work there is still a problem with your esp-idf folder and you should delete it and start over from scratch.

teixeluis, 2020-08-09:
Ok, cloned again esp-idf and started from scratch, deleting the build directory in esp32-cam, and also running ` idf.py fullclean` just in case. Now it went much further, building the bootloader. It just failed later in ldgen:

```
[62/62] Generating binary image from built executable
esptool.py v2.8
Generated /home/lteixeira/projectos/esp32/esp32-cam/build/bootloader/bootloader.bin
[855/858] Generating ld/esp32.project.ld
FAILED: esp-idf/esp32/ld/esp32.project.ld 
cd /home/lteixeira/projectos/esp32/esp32-cam/build/esp-idf/esp32 && /home/lteixeira/.espressif/python_env/idf4.0_py3.5_env/bin/python /home/lteixeira/projectos/esp32/esp/esp-idf/tools/ldgen/ldgen.py --config /home/lteixeira/projectos/esp32/esp32-cam/sdkconfig --fragments /home/lteixeira/projectos/esp32/esp/esp-idf/components/esp_ringbuf/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/spi_flash/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/esp_wifi/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/lwip/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/esp_event/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/espcoredump/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/esp32/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/esp32/ld/esp32_fragments.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/xtensa/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/soc/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/heap/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/freertos/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/newlib/esp32-spiram-rom-functions-c.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/newlib/newlib.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/app_trace/linker.lf	/home/lteixeira/projectos/esp32/esp/esp-idf/components/esp_gdbstub/linker.lf --input /home/lteixeira/projectos/esp32/esp/esp-idf/components/esp32/ld/esp32.project.ld.in --output /home/lteixeira/projectos/esp32/esp32-cam/build/esp-idf/esp32/ld/esp32.project.ld --kconfig /home/lteixeira/projectos/esp32/esp/esp-idf/Kconfig --env-file /home/lteixeira/projectos/esp32/esp32-cam/build/config.env --libraries-file /home/lteixeira/projectos/esp32/esp32-cam/build/ldgen_libraries --objdump
usage: ldgen.py [-h] [--input INPUT] [--fragments FRAGMENTS [FRAGMENTS ...]]
                [--libraries-file LIBRARIES_FILE] [--output OUTPUT]
                [--config CONFIG] [--kconfig KCONFIG] [--env NAME=VAL]
                [--env-file ENV_FILE] [--objdump OBJDUMP]
ldgen.py: error: argument --objdump: expected one argument
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
Cheers

teixeluis, 2020-08-09:
The initial error returned after recreating sdkconfig (deleted the file and did another run of idf.py menuconfig). Tried disabling in menuconfig the elliptic curve ciphers and in the supplicant section, the use of mbedtls, but the error persists.

Cheers

teixeluis, 2020-08-09:
The release/v4.0 still actively receives commits. Can it be the case that there is a regression in their code, given these more recent commits? @bkeevil is there a specific tag or commit # in the esp-idf project that you suggest checking against?

Thanks

Cheers

bkeevil, 2020-08-09:
This is the one I have checked out:
```
commit 4c81978a3e2220674a432a588292a4c860eef27b (HEAD, tag: v4.0.1, origin/release/v4.0)
```

bkeevil, 2020-08-09:
Try using make instead of idf.py to see if it is a problem with your python configuration

```
make menuconfig
make app
make flash
```
Also, there should be some linker executables at /home/lteixeira/.espressif/ that may be a different version. Perhaps delete that directory and run the esp-idf install.sh again. I recall that install.sh installs those.

teixeluis, 2020-08-09:
Hi @bkeevil, via make it compiled just fine, and was also able to flash to the device. 

Now I am having another problem, but probably it is just me (I didn't define an default AP where to connect to in the configuration, and the device doesn't seem to be starting up in AP mode - no  SSID called "ESP32-Camera" shows up).

Should we expect it to start in AP mode and present a captive portal for entering/configuring our network details in runtime?

Thanks

Cheers

bkeevil, 2020-08-09:
It did work at one point but may have been broken by some recent contributions that I was unable to test.

teixeluis, 2020-08-12:
I will close this ticket, as proceeding directly with make commands worked well in my linux setup. It may be relevant to include this in the documentation, for other users.

Another thing that I would challenge, is if it would make sense to include precompiled binaries, possibly once the captive portal issue is fixed. Before then it can be a thing, because there is likely no other practical way to configure the wifi settings.

Thanks

Cheers.

closed 2020-08-12 by teixeluis