bkeevil/esp32-cam #29: ESP32-CAM doesn't associate to WiFi AP

teixeluis, 2020-08-10:
Hello,

I have defined my home WiFi details via menuconfig, prior to building the firmware image. While running the device, it never seems to associate to the WiFi AP (cannot see it obtain a DHCP lease) nor fallback to its own AP mode. I have double checked the WiFi details and these are correct. Looking at the debug output from the serial port, it seems as if it is not assuming the values defined during menuconfig (blank SSID and password):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7440
ho 0 tail 12 room 4
load:0x40078000,len:15596
load:0x40080400,len:4976
entry 0x40080704
I (70) boot: Chip Revision: 1
I (70) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0.1-378-g26d7fb774 2nd stage bootloader
I (41) boot: compile time 23:32:23
I (41) boot: Enabling RNG early entropy source...
I (47) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 80MHz
I (56) boot: SPI Mode       : QIO
I (60) boot: SPI Flash Size : 4MB
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 factory          factory app      00 00 00020000 00370000
I (83) boot:  1 fonts            WiFi data        01 02 00390000 00040000
I (90) boot:  2 nvs              WiFi data        01 02 003d0000 00004000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2467c (149116) map
I (158) esp_image: segment 1: paddr=0x000446a4 vaddr=0x3ffb0000 size=0x03be8 ( 15336) load
I (163) esp_image: segment 2: paddr=0x00048294 vaddr=0x40080000 size=0x00400 (  1024) load
I (165) esp_image: segment 3: paddr=0x0004869c vaddr=0x40080400 size=0x07974 ( 31092) load
I (184) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x908d4 (592084) map
I (343) esp_image: segment 5: paddr=0x000e08f4 vaddr=0x40087d74 size=0x14cbc ( 85180) load
I (389) boot: Loaded app from partition at offset 0x20000
I (389) boot: Disabling RNG early entropy source...
I (389) psram: This chip is ESP32-D0WD
I (394) spiram: Found 64MBit SPI RAM device
I (398) spiram: SPI RAM mode: flash 80m sram 80m
I (404) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (411) cpu_start: Pro cpu up.
I (415) cpu_start: Application information:
I (419) cpu_start: Project name:     esp32-cam
I (424) cpu_start: App version:      86ff7d2
I (429) cpu_start: Compile time:     Aug 10 2020 23:27:09
I (436) cpu_start: ELF file SHA256:  1dd78a5e131ef244...
I (441) cpu_start: ESP-IDF:          v4.0.1-378-g26d7fb774
I (448) cpu_start: Starting app cpu, entry point is 0x40081594
I (0) cpu_start: App cpu up.
I (948) spiram: SPI SRAM memory test OK
I (949) heap_init: Initializing. RAM available for dynamic allocation:
I (949) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (955) heap_init: At 3FFBCA60 len 000235A0 (141 KiB): DRAM
I (961) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (967) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (974) heap_init: At 4009CA30 len 000035D0 (13 KiB): IRAM
I (980) cpu_start: Pro cpu start user code
I (985) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1005) spi_flash: detected chip: generic
I (1005) spi_flash: flash io: qio
I (1005) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1073) settings: NVS Flash Init
I (1073) settings: Settings loaded from NVS
I (1073) settings:  hostname=espressif
I (1073) settings:  wifi_ssid=
I (1073) settings:  wifi_password=
I (1083) settings:  mdns_instance=ESP32 Camera Web Server
I (1093) settings:  ntp_server=pool.ntp.org
I (1093) settings:  timezone=
I (1093) settings:  dhcp=1
I (1103) settings:  ip=0.0.0.0
I (1103) settings:  netmask=0.0.0.0
I (1103) settings:  gateway=0.0.0.0
I (1113) settings:  dns1=0.0.0.0
I (1113) settings:  dns2=0.0.0.0
I (1123) sccb: pin_sda 26 pin_scl 27

I (1123) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1263) camera: Detected OV2640 camera
I (1263) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1263) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1273) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1283) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1293) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1303) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1313) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1313) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1323) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1333) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1343) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1353) camera: Allocating 2 frame buffers (117 KB total)
I (1363) camera: Allocating 58 KB frame buffer in OnBoard RAM
I (1363) camera: Allocating 58 KB frame buffer in OnBoard RAM
I (1553) wifi:wifi driver task: 3ffdb220, prio:23, stack:6656, core=0
I (1553) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1553) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1573) wifi:wifi firmware version: 9156a0d
I (1573) wifi:config NVS flash: enabled
I (1573) wifi:config nano formating: disabled
I (1573) wifi:Init data frame dynamic rx buffer num: 32
I (1583) wifi:Init management frame dynamic rx buffer num: 32
I (1583) wifi:Init management short buffer num: 32
I (1593) wifi:Init dynamic tx buffer num: 32
I (1593) wifi:Init static rx buffer size: 1600
I (1603) wifi:Init static rx buffer num: 10
I (1603) wifi:Init dynamic rx buffer num: 32
I (1613) wifi station: Connecting to AP SSID: password:
I (1613) wifi station: wifi_init_sta finished.
I (1623) wifi station: connecting to ap SSID: password:
I (1703) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1703) wifi:mode : sta (98:f4:ab:00:8f:98)
I (1703) wifi:set country: cc= schan=1 nchan=11 policy=0
```
Is there anything else that needs to be done or configured, in order to make sure that the WiFi credentials are stored in the device (if it is the case that these are not being passed during the build)?

Adding the sdkconfig file.

[sdkconfig.txt](https://github.com/bkeevil/esp32-cam/files/5053796/sdkconfig.txt)

Cheers

bkeevil, 2020-08-10:
Looks like its loading blank strings from NVS. Uncomment lines 47 & 48 of app_main.c to reset NVS. That should cause whatever defaults you have in sdkconfig to be saved to NVS at next bootup. You'll then need to comment those lines out again to be able to change those settings from the web interface.

teixeluis, 2020-08-10:
Ok thanks. Would it make sense to first read the nvs and check the variables, initializing it with values from the firmware (the two commented lines) in case these are blank, and otherwise not touching these? It would eventually have the side-effect of loading the defaults if we would set the SSID and password as blank strings though..

Cheers

teixeluis, 2020-08-10:
Now it registered to the network and is working as expected.

Thanks!

Cheers

bkeevil, 2020-08-11:
Yeah I'll do that

teixeluis, 2020-08-12:
I will close this ticket, as the obstacle was solved by proceeding as you suggested.

Thanks

Cheers

closed 2020-08-12 by teixeluis