bkeevil/esp32-cam #30: Addition of MQTT support

teixeluis, 2020-08-12:
I'm sure this is a topic that may have came across the mind of several people looking into this and similar projects: would you consider relevant to enable MQTT for controlling at least the existing features. E.g.:

 * Control the stream start/stop;
 * Control the camera settings (brightness, contrast, etc);
 * Turn the illuminator led on or off;
 * Considering your wish-list of adding motion detection, publish a message to a configurable topic when motion is detected. Optionally add the URL of the captured image, to the MQTT message payload;
 * Periodic status message with some system information such as Wifi RSSI, memory and cpu usage, esp32 chip internal temperature, etc;
 * Reboot the camera;

Just looking for some feedback if this could make sense.

Thanks.

Cheers

bkeevil, 2020-08-12:
As long as its optional at compile time I don't object. I won't be writing it though.

teixeluis, 2020-08-12:
Fair enough :)