bkeevil/esp32-cam #34: Support for IR motion sensor

mase76, 2021-01-24:
Hi!
Will there also be support for an IR motion sensor?

bkeevil, 2021-02-11:
Its a possibility. Not a lot of free memory on these devices so all additional features should be something users enable in firmware via `make menuconfig`.

g8ecj, 2021-05-29:
I'd be interested in motion detection (no need for IR) with the following main loop:
 * sleep for 500ms
 * grab frame (& decimate down to reasonable pixel count)
 * compare to last frame
 * if large differences then fire up wifi, stream a few seconds worth at a server
 * shut down wifi (power save)
 * save last frame of burst for next sleep/compare cycle
 The idea for this is low average power to run from a single 18650 Lion cell + solar panel (0.5 to 1w panel)

melvinisken, 2021-06-08:
You could check out https://github.com/alanesq/CameraWifiMotion 
Maybe this has some of the features you are looking for.. I think it is not optimized for power save though..