bkeevil/esp32-cam #35: Program stops responding after I2C errors

camielverdult, 2021-02-15:
The firmware crashes/hangs after spitting out ~2 minutes of `i2c: i2c_set_pin(825): scl and sda gpio numbers are the same`. I suspect this has to do with [this](https://github.com/espressif/esp-idf/issues/6293#issuecomment-747817518). I do not know where to implement this, otherwise I would. More log: ```
I (1598) sccb: pin_sda 26 pin_scl 27

E (1598) i2c: i2c_param_config(644): i2c clock choice is invalid, please check flag and frequency
E (2618) i2c: i2c_set_pin(825): scl and sda gpio numbers are the same
E (3618) i2c: i2c_set_pin(825): scl and sda gpio numbers are the same
```

BugerDread, 2021-03-15:
you need to add  "conf.clk_flags = 0;" into components/esp32-camera/driver/sccb.c line 58, the whole init function will look like:

```
int SCCB_Init(int pin_sda, int pin_scl)
{
    ESP_LOGI(TAG, "pin_sda %d pin_scl %d\n", pin_sda, pin_scl);
#ifdef CONFIG_SCCB_HARDWARE_I2C
    //log_i("SCCB_Init start");
    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = pin_sda;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_io_num = pin_scl;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = SCCB_FREQ;
    conf.clk_flags = 0;

    i2c_param_config(SCCB_I2C_PORT, &conf);
    i2c_driver_install(SCCB_I2C_PORT, conf.mode, 0, 0, 0);
#else
    twi_init(pin_sda, pin_scl);
#endif
    return 0;
}
```

BugerDread, 2021-03-15:
or simply update the camera driver, is fixed in master, they zeroed-out the config using memset first

PR #36 (2021-03-19): I2C error and WiFi settings fixed
updated camera driver to lastest - this fix #35 
fixed wifi settings to 
- allow all channels 1-13 (instead 1-11)
- connect to the strongest AP in case you have more APs with same SSID/pwd (scan_method = WIFI_ALL_CHANNEL_SCAN) - see https://github.com/espressif/arduino-esp32/pull/4947 or https://github.com/espressif/esp-idf/issues/6595 for details
closed 2021-03-19 by bkeevil