bkeevil/esp32-cam #39: Header fields are too long for server to interpret

Kasperdelasopa, 2021-03-30:
I'm barely able to connect to the esp32-cam

in 90% there is the error "Header fields are too long for server to interpret". The error is also visible on the Serial port.

Is there a configuration i might missed?

Best regards

bkeevil, 2021-03-30:
I have not heard of that error before. Can you see if you get this error with the version of the code base prior to the last pull request that was merged:

 `git checkout 40299202dc5105646c6cb2937a2e8610dd6f75a2`

Kasperdelasopa, 2021-03-31:
I tried that version --> Same result

BUT: I found a configuration for the HTTP header length in the which seems to do the job. 

Know it seems i have a hardware problem which is not part of this issue.

--> THX

closed 2021-03-31 by Kasperdelasopa