bkeevil/esp32-cam #41: esp32 cam has an error after the flash is successful

adi160885, 2021-09-10:
may i ask for help sir, my esp32 cam error repeatedly after the flash is successful
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
ets Jun  8 2016 00:22:57

bkeevil, 2021-09-10:
It doesn't look like debug logging is turned on

adi160885, 2021-09-13:
it seems i am using version ( ESP IDF 4.3 ) is it correct i did this error

bkeevil, 2021-09-13:
It was pretty finicky about which ESP-IDF version you use. Maybe try using the 4.0 branch that the code was last tested on. There are probably some minor modifications required to get it to run on 4.3 if you are up for it.

adi160885, 2021-09-13:
thank you sir, but I hope it can be used in (ESP-IDF 4.3) maybe you can tell us how to modify it because I am a beginner who is still learning, thank you in advance.